exelban/stats #324: Feature request: Add less bright frames and colours

nikodiyana, 2021-01-28:
Great app!

It will be nice to have not so bright white colours of the frames and boxes, as they take the attention of the colour bars.
Some transparency or greyish colour will be great.

Additional suggestion: It will be great to have an option for brightness control of colours for more subtle colours in general. When working on photos on a dark screen the colour markers of the CPU, RAM etc are very bright and distracting even when chosen less saturated colours.

Keep up the good work!

exelban, 2021-01-31:
Hi. Thanks for your feedback.
If you don't like boxes or frame around the widget, you can always disable it)
I add them for one reason: to make the widget more visible and clear to read. And I'm not sure that making them transparent or greyish it's ok for this reason.

I think the app will receive an option to create your own color palette for widgets)

nikodiyana, 2021-02-01:
Hi, thanks for your attention on the subject!

Yes, indeed, a custom colour system would be perfect. Looking forward to it.

One suggestion: an xml file containing colours would be OK for start, a colour selection UI can be written later.
Stats is mostly used by techies and power users, so editing a text file with settings wouldn't be a problem for them. Also xml can be reused between OS re-installs and shared between users.

Best!

exelban, 2021-02-01:
I will not add any xmls for sure)
Why? Because not everyone know what it is and how they can use it. Even if 99% users of app are `geeks`. I want to have easy tool to monitoring the macOS. Not only a `geek` tool.

I have for a long time an idea to make settings sharable. For sure not throw files, but cloud.

closed 2021-02-07 by exelban