exelban/stats #325: Disk sensor does not update until restart

ivan-avalos, 2021-01-29:
**Describe the bug**
Disk sensor does not update until Stats is restarted.

**Details:**
 - Device: MacBook Air 2017.
 - macOS 11.1
 - Application version: 2.4.14

exelban, 2021-01-29:
Hi. Thanks for reporting.
What do you mean by `Disk sensor`?
Disk usage does not update? Or some sensor?

ivan-avalos, 2021-01-29:
I updated the term in the title and description to "disk usage".

exelban, 2021-01-30:
Could you please provide a little bit more details?
It works and stops working at some moment? Or it doesn't work from the start.
How you noticed that it stops working?

ivan-avalos, 2021-01-30:
It simply does not update, like, at all. At start, it shows the correct disk space, but it never updates until I quit and restart Stats. It used to work, but some update recently caused it to stop working. Everything else still works fine, though.

exelban, 2021-01-31:
Could you check what update interval has the Disk module?

exelban, 2021-01-31:
Don't answer the last question. I find the bug.

closed 2021-01-31 by exelban