exelban/stats #326: Visual bug of disk usage indicator

MasoodFallahpoor, 2021-01-30:
Sometimes the "Disk usage" indicator is displayed wrongly. If I close Stats and open it again, the bug goes away. The following image shows the bug.

[image]

 - Device: MacBook Pro 2019
 - MacOS Catalina 10.15.7
 - Application version: 2.4.14

exelban, 2021-01-30:
Hi. Thanks for reporting. 
I saw it from time to time. It's some bug in the macOS(
It will go to the norm on the next update of the widget.

exelban, 2021-01-31:
This bug related not only to the Disk widget. But to all text widgets.

closed 2021-02-21 by exelban