exelban/stats #327: Obtain human-readable descriptions for SMC keys using /usr/lib/libSMC.dylib

yujitach, 2021-01-31:
At present human-readable descriptions of  SMC keys are hard-coded in [Modules/Sensors/values.swift](https://github.com/exelban/stats/blob/master/Modules/Sensors/values.swift) , but I recently learned that those descriptions can be obtained programmatically using `/usr/lib/libSMC.dylib`, as demonstrated in [`smc_dump_lib_keys`](https://github.com/acidanthera/VirtualSMC/blob/4411a1aa53c147928c406d72f3e2035207e0d1aa/Tools/smcread/smcread.c#L516)  in `smcreader.c` in [VirtualSMC](https://github.com/acidanthera/VirtualSMC/).

I think this method is more future-proof. 

( I found this info while maintaining my [MenuMeters](https://github.com/yujitach/MenuMeters/), which is basically an older Obj-C version of your excellent app, which is basically everything what I wanted my app to become in the far future. Great job!)

exelban, 2021-01-31:
Hi. Thanks.
I will take a look at this method. I think it could be very useful for m1 (#210).

Btw, thanks for maintaining MenuMeters. I was using your fork for a while)

yujitach, 2021-01-31:
It should be useful for Apple Silicon, but the data type is changed from `sp78` to `sp7s` and/or `flt `. The latter is simply a `float` but I haven't figured out what is `sp7s` yet. For now I'm not using SMC on Apple Silicon. Instead I'm using the code you complained about in https://github.com/exelban/stats/issues/210 . See [my version here](https://github.com/yujitach/MenuMeters/blob/master/hardware_reader/applesilicon_hardware_reader.m).

The code base of MenuMeters shows its age. It's written in a very old-fashioned way and very difficult to maintain. I'm glad that you started a total rewrite by Swift.

exelban, 2021-04-11:
Hi. Thanks for these tips. I transformed your method for Apple Silicon to [swift](https://github.com/exelban/stats/blob/2528830a148e1c061daa908bc61fcd00157738de/Modules/Sensors/readers.swift#L126-L199).

I also take a look at the `libSMC.dylib` library. But have no found new names for SMC sensors. Also, `libSMC.dylib` is not available on macos 11(

closed 2021-04-11 by exelban