exelban/stats #329: Program displays only one GPU – Mac Pro (2013)

molekulek, 2021-01-31:
I haven't seen the second graphic info for several versions of the program. 

Nazwa modelu:	Mac Pro (2013)
Identyfikator modelu:	MacPro6,1
2x AMD FirePro D500

[image]

[image]

exelban, 2021-01-31:
Hi. Thanks for reporting.

It's because both GPUs using the same drivers. And my app trying to detect graphics by its driver. I forgot that there is a mac with 2 identical GPUs)

I will try to fix it.

exelban, 2021-01-31:
Could you please provide the output of `sysctl machdep.cpu.brand_string`?
I will try to fix the CPU name. `v22.50GHz` - there must be a space)

molekulek, 2021-02-01:
Mac-Pro:~ $ sysctl machdep.cpu.brand_string
machdep.cpu.brand_string: Intel(R) Xeon(R) CPU E5-1650 v2 @ 3.50GHz

In previous versions of stats (~ three months ago) there were two graphics cards, but they were badly recognized, now they are well recognized, but there is only one in the monitor.

Thanks for developing the program!

closed 2021-02-21 by exelban