exelban/stats #332: Apple Silicon - Monitor video decoding engine

singhkays, 2021-02-02:
From the M1 block diagram it has a video decode engine for supported video formats. Is it possible to monitor the usage of this engine? 

> [image]

This would be similar to the video decode monitoring in Windows task manager
> [image]

exelban, 2021-02-02:
Hi.

I'm not sure that macOS provide some sort of API to get usage of this engine. Also, I'm pretty sure that it's not a dedicated chip. But a part of SoC (as Neural engine) or GPU. If yes, it will be impossible. 

I don't have m1 to check what I can do with this.

singhkays, 2021-02-02:
@exelban Got it. I have a M1 Mac Mini. If you want me to run some tests, I'm happy to.

exelban, 2021-02-02:
I'm not even sure what you can test(
First of all, need to find some information about this chip. It's a dedicated chip or part of SoC. And what API Apple provides to it.

singhkays, 2021-02-04:
Some info from FFmpeg - https://trac.ffmpeg.org/wiki/HWAccelIntro#VideoToolbox

and from Apple - https://developer.apple.com/documentation/videotoolbox

exelban, 2021-02-07:
Ok. I will not spend time on this thing. I suppose there are not many people which really need this. Also, I do not have M1 to check out if I can read `decoding engine`. If you want to add this feature to the app, PR is welcomed. Otherwise, issue can be closed.

singhkays, 2021-02-08:
No worries, I understand. Closing for now.

closed 2021-02-08 by singhkays