exelban/stats #333: issue partition type

sam-anam1, 2021-02-02:
Hello and hope you are fine as well

with NTFS and ExFAT Volumes not showing used and free space. 
Version 2.4.15

I have attached screenshot
(https://user-images.githubusercontent.com/6969182/106640966-bbca5a80-6597-11eb-8f4c-187cdd4b8ea1.png)

exelban, 2021-02-02:
Hi. Thanks for reporting.
I will take a look at this.

exelban, 2021-02-07:
Hi. I make some tests with ExFAT and it works fine. NTFS will not work, because macOS does not have drivers for it.

closed 2021-02-07 by exelban

sam-anam1, 2021-02-07:
thank you so much!
at the point when first installed it was on in November versions it think it was all working drives showing right 
os is Big Sur

sam-anam1, 2021-02-15:
hello @exelban 
I can I confirm that I just installed random version Stats 2.4.9v it works well and shows NTFS n' ExFAT
I just downloaded and replace with the last version its working and install back the last ver. it dose not  show