exelban/stats #334: Network widget - hovering over IP address shows wrong mouse cursor style

JeffreyCA, 2021-02-02:
**Describe the bug**
When you hover over the IP address in the network widget, the mouse cursor changes from:

This:

[image]

to this:

[image]

to this:
[image]

I think it should always show the hand cursor and not show the I-beam cursor. 

**Details:**
 - Device: Macbook Pro 2019
 - MacOS 11.2
 - Application version: 2.4.15

exelban, 2021-02-02:
Hi.
I will take a look at this strange behavior.

exelban, 2021-02-08:
Hi. 
I have removed the arrow and edit pointers. Only a normal one leaves. But it still copyable. It's a temporary solution. It will be completely filled with #339.

closed 2021-02-14 by exelban

JeffreyCA, 2021-02-14:
Thanks, can confirm it's fixed now. It would be nice to also have some visual feedback to indicate the address has been copied to the clipboard.