exelban/stats #335: Stats 2.4.14 Catalina 10.15.7 crash with EXC_BAD_INSTRUCTION

mivens, 2021-02-03:
**Describe the bug**
Stats version 2.4.15 has crashed twice in eu.exelban.Stats.Disk. Trigger is unknown - the Mac was unused with the screen locked in the middle of the night for one of the crashes.
[Stats.crash.tar.gz](https://github.com/exelban/stats/files/5917380/Stats.crash.tar.gz)

From the first crash file:

```Process:               Stats [1234]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.4.15 (192)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Stats [1234]
User ID:               502

Date/Time:             2021-02-03 04:10:30.309 +0000
OS Version:            Mac OS X 10.15.7 (19H512)
Report Version:        12
Anonymous UUID:        42E538ED-D7B0-02FE-F1DF-7B333613F0EB

Time Awake Since Boot: 32000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [1234]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   eu.exelban.Stats.Disk               0x000000010d957c45 0x10d94e000 + 40005
1   eu.exelban.Stats.Disk               0x000000010d9578ee 0x10d94e000 + 39150
2   eu.exelban.Stats.Disk               0x000000010d9550f8 0x10d94e000 + 28920
3   eu.exelban.Stats.Disk               0x000000010d954e26 0x10d94e000 + 28198
4   eu.exelban.Stats.Disk               0x000000010d95371b 0x10d94e000 + 22299
5   eu.exelban.Stats.Disk               0x000000010d95e95e 0x10d94e000 + 67934
6   libdispatch.dylib                   0x00007fff6fa606c4 _dispatch_call_block_and_release + 12
7   libdispatch.dylib                   0x00007fff6fa61658 _dispatch_client_callout + 8
8   libdispatch.dylib                   0x00007fff6fa6ccab _dispatch_main_queue_callback_4CF + 936
9   com.apple.CoreFoundation            0x00007fff35bf2e81 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
10  com.apple.CoreFoundation            0x00007fff35bb2c87 __CFRunLoopRun + 2028
11  com.apple.CoreFoundation            0x00007fff35bb1e3e CFRunLoopRunSpecific + 462
12  com.apple.HIToolbox                 0x00007fff347deabd RunCurrentEventLoopInMode + 292
13  com.apple.HIToolbox                 0x00007fff347de7d5 ReceiveNextEventCommon + 584
14  com.apple.HIToolbox                 0x00007fff347de579 _BlockUntilNextEventMatchingListInModeWithFilter + 64
15  com.apple.AppKit                    0x00007fff32e24039 _DPSNextEvent + 883
16  com.apple.AppKit                    0x00007fff32e22880 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
17  com.apple.AppKit                    0x00007fff32e1458e -[NSApplication run] + 658
18  eu.exelban.Stats                    0x000000010d9175ed 0x10d914000 + 13805
19  libdyld.dylib                       0x00007fff6fabacc9 start + 1
```
Second crash file shows similar:

```Process:               Stats [18388]
Path:                  /Applications//*/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.4.15 (192)
Code Type:             X86-64 (Native)
Parent Process:        ??? [18383]
User ID:               502

Date/Time:             2021-02-01 11:17:36.551 +0000
OS Version:            Mac OS X 10.15.7 (19H114)
Report Version:        12
Anonymous UUID:        42E538ED-D7B0-02FE-F1DF-7B333613F0EB

Time Awake Since Boot: 920000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [18388]

Application Specific Information:
dyld: in dlopen_preflight()

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   eu.exelban.Stats.Disk               0x000000010d5277a0 0x10d51e000 + 38816
1   eu.exelban.Stats.Disk               0x000000010d528d33 0x10d51e000 + 44339
2   eu.exelban.Stats.Disk               0x000000010d525217 0x10d51e000 + 29207
3   eu.exelban.Stats.Disk               0x000000010d524e26 0x10d51e000 + 28198
4   eu.exelban.Stats.Disk               0x000000010d52371b 0x10d51e000 + 22299
5   eu.exelban.Stats.Disk               0x000000010d52e95e 0x10d51e000 + 67934
6   libdispatch.dylib                   0x00007fff6a7e86c4 _dispatch_call_block_and_release + 12
7   libdispatch.dylib                   0x00007fff6a7e9658 _dispatch_client_callout + 8
8   libdispatch.dylib                   0x00007fff6a7f4cab _dispatch_main_queue_callback_4CF + 936
9   com.apple.CoreFoundation            0x00007fff309b1e81 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
10  com.apple.CoreFoundation            0x00007fff30971c87 __CFRunLoopRun + 2028
11  com.apple.CoreFoundation            0x00007fff30970e3e CFRunLoopRunSpecific + 462
12  com.apple.HIToolbox                 0x00007fff2f59dabd RunCurrentEventLoopInMode + 292
13  com.apple.HIToolbox                 0x00007fff2f59d7d5 ReceiveNextEventCommon + 584
14  com.apple.HIToolbox                 0x00007fff2f59d579 _BlockUntilNextEventMatchingListInModeWithFilter + 64
15  com.apple.AppKit                    0x00007fff2dbe3039 _DPSNextEvent + 883
16  com.apple.AppKit                    0x00007fff2dbe1880 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
17  com.apple.AppKit                    0x00007fff2dbd358e -[NSApplication run] + 658
18  eu.exelban.Stats                    0x000000010d4e55ed 0x10d4e2000 + 13805
19  libdyld.dylib                       0x00007fff6a842cc9 start + 1
```

**Details:**
 - Device: MacBook Pro (Retina, 13-inch, Early 2015)
 - MacOS 10.15.7
 - Application version 2.4.15

exelban, 2021-02-03:
Hi. Thanks for reporting.
I will take a look later totay.

exelban, 2021-02-03:
There is a build with a fix. You can use it till the new release.

[Stats.dmg.zip](https://github.com/exelban/stats/files/5920357/Stats.dmg.zip)

closed 2021-02-07 by exelban