exelban/stats #336: Turn off Battery must hot down

Leo7Chuang, 2021-02-03:
Hi! 
I'm from Taiwan and I'm sorry about I'm not good for English.
When I want to turn off Battery, Stats Must show down. I can't use Force Quit to quit it. I quit it via Activity Monitor only.

This is big problem I guess, btw this is more beautiful than iStats. I like it!!
Hope this bug can be fix soon.

Leo Chuang

exelban, 2021-02-03:
Hi.

I'm not sure what you trying to achieve. But if you want to disable the battery module you can do it in the settings:
[image]

Leo7Chuang, 2021-02-03:
Yes! After I turn it off, and the app must crash!

RajceP, 2021-02-03:
I face the same bug. When i disable battery widget the app crashes.

Leo7Chuang, 2021-02-03:
I apologize for my bad English skill!

exelban, 2021-02-03:
Could you please post the crash report?

You can find a crash report in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

RajceP, 2021-02-03:
It seems it does't crash completely, rather it s in the loop (Beach Ball of Death). Theres no crash report. I have to manually kill the app.

Leo7Chuang, 2021-02-03:
I can’t find it in Crash Reports.
Sorry!

exelban, 2021-02-03:
Hmm, strange. Ok, I will try to debug it. Thanks.

exelban, 2021-02-03:
Just to be sure. Does it happen when you disabling the battery module?
Could you try to disable another module?
After the restart, does the battery module enabled again?

RajceP, 2021-02-03:
Yes.
Only battery module, tried all of them.
After restart, battery module is disabled.

Leo7Chuang, 2021-02-03:
Battery module can be enabled successfully.
Other module can be enabled and disabled successfully.

Ohh... I find the another situation.
I find it in Activity Monitor was not responding after app crash for almost 1 minute.
It isn't immediate to show it is't responding.

liuzy0404, 2021-02-05:
Same bug, when switch the Battery label off. Console.app doesn't have crash log.

My environment:
MacBook Pro (13-inch, 2016, Four Thunderbolt 3 Ports)
Big Sur 11.2
Stats 2.4.15

closed 2021-02-07 by exelban