exelban/stats #344: Auto update doesn't work properly with Stats.app placed in /Applications/Utilities

wzxu, 2021-02-14:
**Describe the bug**
I prefer putting apps that I consider "utilities" not under /Applications, but, well, /Applications/Utilities. 
What I've been noticing is that Stats' auto update always fails silently with an invalid app left in its place.
[image]
Thinking it might be a permission problem. I didn't report the bug before.
But today I realized that the downloaded updated app is actually intact and executable, just placed in the wrong place!
In short, this is the outcome:
- /Applications/Utilities/Stats.app → Original app; now invalid
- /Applications/Utilities/Stats.app/Stats.app → Misplaced updated app
[image]

**Details:**
 - MacOS 11.2.1
 - Application version 2.5.0

exelban, 2021-02-14:
Hi. Thanks for reporting.
I will try to fix it.

Jethro1000, 2021-02-16:
/Applications/Utilities is a reserved directory for Apple utilities. Since macOS 10.6 it should not used for customer installed programs. If you want use this directory, it could be possible that programs which are not Apple installed are removed after re-install macOS vi recovery

wzxu, 2021-02-16:
@Jethro1000 Thanks for the input. I did not realize there's such limitation on that directory, and have been putting apps there since forever (even directed some of my Homebrew casks installs there in my scripts/dotfiles for setting up new machines).

Still, I think this is a pretty curious bug and it's greatly appreciated if fixable. If not then I can just put it back to /Applications and call it a day, so please take your time @exelban :)

exelban, 2021-02-17:
Hi. I take a look at this behavior. And there is a problem with permissions. Because apps in the `/Applications/Utilities` require `sudo` to be updated (added/replaced/deleted). 
I have no idea how it can be fixed. Only if check if an app installed in the `Applications/Utilities`. If yes -> require a password. But I don't like this idea.

wzxu, 2021-02-18:
As @Jethro1000 said, it seems this directory was not meant for user apps. I always change the permission of this folder when I set up my new Macs so permission is not a problem for me. I was mainly reporting for the part where the updated app gets placed within itself.

PS. I notice normal apps that update via Sparkle does automatically ask for password when it doesn't have sufficient permission, but since Stats seems to use its own update mechanism, I think the permission part may not be easily handled. Again, it's not the main reason I reported this issue, so it doesn't need to be addressed here. Thanks.

wzxu, 2021-03-01:
Updated from v2.5.1 to v2.5.3 via auto update successfully. Confirmed fixed.

closed 2021-03-01 by wzxu