exelban/stats #346: After the latest update no stats are showing on the menu bar

simonerancati, 2021-02-14:
After updating to version 2.5.0 on my M1 MacBook Air with Big Sur 11.2.1 nothing is showing on the menu bar

[image]

JJvanRoest, 2021-02-14:
Same on 15" MacBook Pro 2018, only battery percentage is showing. Happens on both Catalina and after updating to Big Sur today.

Tai-Zhou, 2021-02-15:
> After updating to version 2.5.0 on my M1 MacBook Air with Big Sur 11.2.1 nothing is showing on the menu bar

In 2.5.0, you can select several widgets at the same time, and it seems you selected none of them in the figure you post. After being selected, a blue box should appear like this.

[image]

mrchoke, 2021-02-15:
Macbook Pro 2020 Intel base 2.5.0 menu bar not showing too. 😭

leifwenzel, 2021-02-15:
Same on Macbook Pro 2017 with macOS 10.15.7
So it is impossible to select any widget, because i can't get to application window to select these.

mrchoke, 2021-02-15:
> Same on Macbook Pro 2017 with macOS 10.15.7
> So it is impossible to select any widget, because i can't get to application window to select these.
> Update:
> I've managed to get to the application window by double clicking again on the application icon. Then the application window appears. Now you have to drag and drop these little widget (with percertage or bars) to your menu bar. Then it works for me.

Thank you @leifwenzel  it work now 😘

igorkulman, 2021-02-15:
I have a related issue, after updating to 2.5.0 on Big Sure 11.2 the Fans, Sensors, Network widgets are not showing when enabled. The others show fine.

simonerancati, 2021-02-15:
> > After updating to version 2.5.0 on my M1 MacBook Air with Big Sur 11.2.1 nothing is showing on the menu bar
> 
> In 2.5.0, you can select several widgets at the same time, and it seems you selected none of them in the figure you post. After being selected, a blue box should appear like this.
> 
> [image]

Thank you so much, when I saw that after the update everything was removed I thought it was a bug

exelban, 2021-02-15:
Hi. Thanks for reporting. As @Tai-Zhou noticed in the new release you can enable few widgets. And it looks like this feature disables some of your widgets. Please make sure that at least 1 widget is enabled.

kauelima, 2021-02-17:
Thanks Guys! This thread worked for me. If you also use Vanilla to hide/show icons, make sure to also open vanilla's preferences because clicking on the sensors you want will do nothing since they will be added to "Permanently Hide" section.

But with those information it all worked!

pasdee, 2021-02-18:
wow thanks!

kauelima, 2021-02-18:
@exelban A suggestion for making this clearer would be using an always visible bar on top with all your selected widget. This would improve greatly the feedback that clicking a widget will select/deselect it (because that click will add/remove the said widget from the bar up top) and also giving a view for users to review all of the widgets they have available at a given moment :)

exelban, 2021-02-18:
@kauelima thanks for propose. I'm on the road to this feature)

Hani-Ghazi, 2021-02-20:
> > After updating to version 2.5.0 on my M1 MacBook Air with Big Sur 11.2.1 nothing is showing on the menu bar
> 
> In 2.5.0, you can select several widgets at the same time, and it seems you selected none of them in the figure you post. After being selected, a blue box should appear like this.
> 
> [image]

Thank you sir! this worked for me after spending like an hour to figure out what happened

closed 2021-02-28 by exelban