exelban/stats #348: Sensors widget width should be fixed

lfom, 2021-02-18:
**Describe the bug**
The change in the width makes the widgets on the left jump around, I think [the dev agrees with me that it is not desirable](https://github.com/exelban/stats/issues/92#issuecomment-698172840). 😋 
Probably two digits for integer part of Voltage and Power, three for temperature and four for the fans are enough (the largest one would make the width fixed, so no complicated logic is needed, I guess).

Quick question: what Unknown was supposed to show? I am running Catalina on an unsupported iMac if it matters. 

Thanks a lot for this incredibly good looking and useful app! I was surprised it uses even less memory than my previous favorite app. 😉 

**Details:**
 - Device: iMac12,1
 - MacOS 10.15.7
 - Application version 2.4.16 (I downgraded from 2.5.0 because of the problem saving the widget order, but it does not affect the issue)

[image]

exelban, 2021-02-18:
Hi. You see `Unknown` because `iMac12,1` was not on the official support list of Catalina. I will add this model to the list.
Later I will take a look at the first bug.

exelban, 2021-02-18:
Ok. About fixed width of Sensors module. I'm not sure if it's possible. Because the Sensors widget has a too big range of width. And if the static width must be the max of all possible. It will be too wide.

lfom, 2021-02-18:
@exelban I see... Maybe a minimum of 4 characters, so it works well most of the times (two digits for integer, plus dot, plus one decimal for voltage and power, it should be good enough for temperature as well.

exelban, 2021-02-21:
Yes. In theory, it could be 4 characters for value + space + 3 characters for symbols (RPM for fan). So the fixed width must be for 8 characters. Which ok for people who use it only for temperature or another sensor which has a lot of chars. But I'm pretty sure that after this change someone will come and ask to make it smaller because there is too much space when showing only the temperature.

lfom, 2021-02-21:
That's why I suggested the minimum for voltage/power/temp. There is another widget for the fans only, right?

exelban, 2021-02-22:
I agree. The widget can have a fixed width If there will be only voltage/power/temp. But #197.
It will not be the settings option for sure.

rozhl, 2021-02-28:
Also there should be an option to remove the "RPM" to make the occupied space smaller.

exelban, 2021-03-28:
There is a lot of problem with this widget. And I will not make it fixed width. At least till it will be rewritten. Also, I'm thinking to remove fans from this module. So maybe it will be much easiest to make it with fixed width.

closed 2021-03-28 by exelban