exelban/stats #351: When fan speed goes to zero, last nonzero value is displayed

davidfokkema, 2021-02-18:
I have the 'fans' widget enabled and it seems to do the trick: displaying the fan speed and updating once per second. However, when my notebook cools down and the fans are turned off the widget keeps displaying static numbers (like 689 / 244 rpm). I'm pretty sure the fans are turned off:
```
$ sudo powermetrics -s smc

**** SMC sensors ****

CPU Thermal level: 0
IO Thermal level: 0
Fan: 0 rpm
CPU die temperature: 54.67 C
CPU Plimit: 0.00
GPU Plimit (Int): 0.00
Number of prochots: 0
```
Restarting the application does display 0 rpm, but when the fans have run for a bit and are off again, the numbers are static and nonzero.

**Details:**
 - Device: Macbook Pro 2016
 - MacOS 11.2.1
 - Application version 2.5.0

exelban, 2021-02-18:
Hi. Thanks for reporting.
I will take a look at this strange behavior.

exelban, 2021-02-20:
Hi.
It's not normal to have 0 RPM on the fan. It more looks like you have some problem with the fan or with SMC. The fan has a minimum speed, and macOS will not go below these values. Only if you use some external tools to force them to 0 RPM (I tried to do it, and it not so easy).

If the fan speed stops updating, and after the restart you see 0 RPM. It for 99% some problem with SMC. Because 0 it's a default value and it's set if could not receive the actual speed from SMC.

davidfokkema, 2021-02-20:
@exelban do you have a reference for that? Apple itself says:

> If the ambient temperature is high, the fans turn on sooner and run faster.

(https://support.apple.com/en-us/HT202179)

That suggests that the fans can indeed be turned off. When they *are* running, I do indeed see minimum speeds around 1250 (left) / 1350 (right) RPM, until it is zero again. I reset my SMC just in case, but I still see this behaviour: my Mac is mostly cool (40 - 50 C) unless I'm not on battery power reading my mail. When plugged in, backblaze, photolibraryd and several more processes are more active and my fan is always running. On battery and only lightly using my Mac, it stays very cool.

If the system reports 0 rpm, why not display 0 rpm?

exelban, 2021-02-20:
There is only one way why fan speed will not change. SMC does not return a value. There is no other option.

exelban, 2021-02-20:
Could you maybe check the fan speed via SMC?

davidfokkema, 2021-02-20:
> Could you maybe check the fan speed via SMC?

The only way I know of how to do that is using the powermetrics command, like so:
```
➜ sudo powermetrics -s smc
Password:
Machine model: MacBookPro13,2
SMC version: 2.37f25
EFI version: 429.1.0
OS version: 20D74
Boot arguments:
Boot time: Sat Feb 20 17:18:53 2021

*** Sampled system activity (Sat Feb 20 19:50:17 2021 +0100) (5014.31ms elapsed) ***

**** SMC sensors ****

CPU Thermal level: 0
IO Thermal level: 0
Fan: 0 rpm
CPU die temperature: 48.34 C
CPU Plimit: 0.00
GPU Plimit (Int): 0.00
Number of prochots: 0
```
And when I stress the machine for a bit and right after the fans first kick in, so temperature is high:
```
➜ sudo powermetrics -s smc
Machine model: MacBookPro13,2
SMC version: 2.37f25
EFI version: 429.1.0
OS version: 20D74
Boot arguments:
Boot time: Sat Feb 20 17:18:53 2021

*** Sampled system activity (Sat Feb 20 19:53:25 2021 +0100) (5009.09ms elapsed) ***

**** SMC sensors ****

CPU Thermal level: 31
IO Thermal level: 0
Fan: 1251 rpm
CPU die temperature: 84.28 C
CPU Plimit: 0.00
GPU Plimit (Int): 0.00
Number of prochots: 0
```

exelban, 2021-02-20:
There is one problem with this method. If will show 0 RPM if fan value will be undefined (the same situation when you restart Stats and see 0 value). You could use [this](https://github.com/hholtmann/smcFanControl/tree/master/smc-command) nice tool to fetch keys. 

```
git clone https://github.com/hholtmann/smcFanControl
cd smcFanControl/smc-command
make
./smc -l | grep Ac
```

davidfokkema, 2021-02-20:
With fans 'off':
```
➜ ./smc -l|grep Ac
  F0Ac  [fpe2]  0.00 (bytes 00 00)
  F1Ac  [fpe2]  0.00 (bytes 00 00)
  LAcN  [ui8 ]  0 (bytes 00)
  MSAc  [fp88]  0.000 (bytes 00 00)
```
With fans on:
```
➜ ./smc -l|grep Ac
  F0Ac  [fpe2]  1228.00 (bytes 13 30)
  F1Ac  [fpe2]  1333.00 (bytes 14 d4)
  LAcN  [ui8 ]  0 (bytes 00)
  MSAc  [fp88]  0.000 (bytes 00 00)
```

exelban, 2021-02-20:
Hmm, and after the change from 0->value or value->0 it does not update in the Stats?

davidfokkema, 2021-02-20:
After 0->value, Stats is updating the fan speed every second and seems to be accurate. Fan speeds change slightly from second to second and hover at around 1250 RPM. Whenever value->0 happens Stats stops updating and the value displayed is somewhere in the hundreds (currently 485 / 589 RPM, but I've seen other values as well). It seems to me that Stats is capturing the fans spinning down but stops updating when the SMC reports 0 RPM. Thanks for looking into this!

exelban, 2021-02-20:
Could you run a debug version of Stats with enabled logs and share the logs?

davidfokkema, 2021-02-20:
> Could you run a debug version of Stats with enabled logs and share the logs?
> To see what is going on inside Stats, and why it does not update properly the speed.

Sure! How do I do that?

exelban, 2021-02-20:
I need 20-30min to make it.

davidfokkema, 2021-02-20:
> I need 20-30min to make it.

Ok! Will probably be tomorrow before I can test it then, so no hurries.

exelban, 2021-02-20:
Thanks. There is a debug version with some logs.
Logs will be saved to the file `log.txt` which could be found in the `/Documents`.

Please run this build and simulate the bug with fan speed. And post the logs after.

[Stats.dmg.zip](https://github.com/exelban/stats/files/6016270/Stats.dmg.zip)

davidfokkema, 2021-02-21:
The log seems to report nil instead of zero, at the same time that smc and powermetrics report zero. Thanks again for taking the time to look into this!

On the 'left side' lines there is a nonprinting character which removes the rest of the line in some editors (like textedit). Using less on the command line it reads fine.

[log.txt](https://github.com/exelban/stats/files/6017502/log.txt)

exelban, 2021-02-21:
Thanks for the logs. I think I know how it can be fixed now.

closed 2021-02-21 by exelban

davidfokkema, 2021-02-21:
Great, thanks!

davidfokkema, 2021-02-22:
Works perfectly for the fan widget, thanks! The sensors widget has options for showing temperature and power, but also for showing fan speed. That still exhibits the same problem. So the fan speed widget reports 0 RPM, but the sensors widget reports an unchanging finite value. Maybe you can apply the same fix there?

exelban, 2021-02-22:
Hi. Yeap. I will make the same fix for Sensors)