exelban/stats #354: Feature Request: Update Information

Tai-Zhou, 2021-02-21:
The update information can help the users know the new features and changes, and issues like #346 can be solved before a new issue is submitted. I suggest to show the update information in the update window below the version numbers.

I think the information can be derived like https://github.com/exelban/stats/blob/4b736700b9471bfc78268bec3ce42b1cd1af6c84/StatsKit/updater.swift#L101 with `jsonArray["body"]`, but in the markdown format. Some efforts might be needed to make it show properly in the update window.

exelban, 2021-02-21:
Hi. Thanks for the proposal.
I think it could be added. But not sure if it helps with a situation like #346. As I understand it happens only when the user removes widgets from the menubar manually.

exelban, 2021-05-15:
Hi. I have added a button to open the new version changelog.
I wanted to render md. But it requires a UI for it. I tried a few UIs, but I'm not good at it. So I decided to add a button.
Give me know if you have an idea of redesign the update window with changelog.

closed 2021-05-16 by exelban

rakleed, 2022-07-31:
@exelban can you take part of the code from https://github.com/sparkle-project/Sparkle to display the UI?