exelban/stats #356: The details

RickyLiu0w0, 2021-02-22:
**Describe the bug**
A clear and concise description of what the bug is.

**Details:**
 - Device: [e.g. Macbook Pro 2016]
 - MacOS [e.g. 10.15.5]
 - Application version [e.g. 2.1.11]

RickyLiu0w0, 2021-02-22:
Sorry, I pressed the commit button carelessly before I finished the issue.

Here is the detail information:

**Describe the bug**
Whatever I select used or unused capacity, it also displays 0%.
[image]

** Details:**

- Device: Macbook Pro 2016
- MacOS 10.15.7
- Application version 2.5.1

exelban, 2021-02-22:
Hi. Thanks for reporting. I will fix it.

closed 2021-02-28 by exelban