exelban/stats #359: Feature request:  monitoring for SSD health

tlove004, 2021-02-23:
It would be nice to have quick health stats for SSDs, such as: 

- total read
- total written
- health (as a percentage)

I typically use smartctl from smartmontools for monitoring my SSDs, but a quick glance in Stats would be great!

Thanks!

exelban, 2021-02-24:
Hi. Thanks for the propose.
I will take a look at this.

exelban, 2021-02-27:
Hi. I take a look at this feature. There is one problem. `smartctl` and `smartmontools` are not default macOS tools.
There is a way to receive this information using default macOS tools?

tlove004, 2021-02-27:
Probably not in a very portable way — I’ve looked through their source code and they basically maintain a dictionary of drive vendors and their associated rated specifications (total writes, etc).  I’m guessing this is the best that could be done, as drives don’t store this information internally, so there would be no way to extract it.

exelban, 2021-02-27:
I will not add a feature that will require installing 3d party tools (IGP it's a little bit different story).
So, I will implement this feature if you will find a good solution with default macOS tools.

exelban, 2021-03-14:
Hi. Reopen the issue if there will be an option to do it with internal tools.

closed 2021-03-14 by exelban