exelban/stats #360: Feature Request:  Battery charged notification

tlove004, 2021-02-23:
Similar to the low level notification, but a 'charged level notification' for when battery is being charged and has reached a provided percentage.

Even though Apple notebooks have very good power management to prevent overcharging, I typically try to cycle my battery between ~20% - ~85% to try to get the most life out of my battery.

Thanks for all your work!

exelban, 2021-02-24:
Hi. Thanks for the propose.
I think it could be added.

tlove004, 2021-02-24:
I've stubbed out changes to get this working locally, and was planning on putting in a PR for it; however, it seems when clicking on menubar items, arbitrary space is added between the icons, until they get moved as far left as possible. 

What's odd is that this occurs when building a clean clone from scratch, as well, so it doesn't seem like it's due to code additions I've made.

tlove004, 2021-02-24:
PR #362 implements this; however, there is an oddity when building and testing locally:  when clicking icons, empty gaps in the taskbar are inserted so that the icons are moved to the left.  This happens when building master from scratch as well as my dev fork, so it doesn't seem to be due to code additions I've made.

exelban, 2021-02-25:
Hi. Thanks for the PR. I will take a look later.

Could you please tell me more about this bug with a local run of the master? What do you mean by `empty gaps` and that icons are moved to the left? Maybe you can share the screenshot?

tlove004, 2021-02-25:
Sure.   This is a fresh (archive) build off the master branch; the only changes I made were setting the build team to my own dev account:  

Fresh launch:
[image]

After one click:
[image]

After second click:
[image]

To clarify, the gaps are not _entirely_ empty -- clicking in those areas highlights an icon's-width area of the taskbar, with no action associated with it: 
[image]

2019 16" MBP, 2.3 i9, 32GB
macOS 10.15.7

exelban, 2021-02-25:
Thanks for sharing.

exelban, 2021-02-25:
Could I ask you to make the pull of master and check if this bug still present?

tlove004, 2021-02-25:
That fixed it 👍

exelban, 2021-02-26:
Thanks for reporting and recheck. It really helps.

closed 2021-02-28 by exelban