exelban/stats #364: Feature request: Monochromatic arrows for network widget

igorkulman, 2021-02-26:
**Describe the bug**
I have all the widgets set to be monochromatic but there is no way to do that with the network. This results in the network widget showing blue and red arrows when I want them display, which spoils the overal design.

[image]

**Details:**
 - MacOS 11.2.2
 - Application version 2.5.1

exelban, 2021-02-26:
Hi. These arrows are for an indication about in/outgoing traffic. They will have no sense if they will be monochromatic.

igorkulman, 2021-02-27:
Actually if there is some incoming or outgoing traffic the numbers are greater than zero right.

exelban, 2021-02-27:
Yes. So if you don't want to have arrows/dots/symbols visualization, you can disable it.

closed 2021-03-02 by igorkulman

JeffreyCA, 2021-03-17:
How about the idea of *flashing* monochromatic arrows? When there is no network traffic, hide the corresponding arrow (upload/download), otherwise flash the monochromatic arrow.

exelban, 2021-03-17:
In theory, yes, it could be monochromatic arrows. And not hiding when no traffic, but changing the gray shade.