exelban/stats #365: Feature request: language switcher inside app

winogradoff, 2021-02-26:
Hi!

Please add a language switcher to the app.

I don’t want to switch the whole system to English, and the Russian translation in the application is not entirely correct and I don’t like it. I would only like to switch this application to English

exelban, 2021-02-26:
Hi. Thanks for the proposal. I think it could be added.
Btw, you can always help by improving existing translation)

webjib, 2021-03-04:
Starting with macOS Catalina 10.15, you can force language for any app. Just go to macOS System Preferences > Language & Region > Apps.

winogradoff, 2021-03-04:
> Starting with macOS Catalina 10.15, you can force language for any app. Just go to macOS System Preferences > Language & Region > Apps.

thanks, it helped me

exelban, 2021-03-05:
@webjib Thanks for this tip. It releases me from implementing this feature)

closed 2021-03-05 by exelban