exelban/stats #368: large margin between the icons

n-osennij, 2021-02-28:
**Describe the bug**
After a while, a large margin appears between the icons by itself  in topbar.

[image]

**Details:**
 - Device:  Air 2020 m1, 8GB ram, 7 GPU
 - MacOS 11.2.1
 - Application version 2.5.2

exelban, 2021-02-28:
Hi. Thanks for reporting. Could you please check if there is something in these empty spaces?
I mean, does it wrong width of each widget, or there is another widget in these spaces?

n-osennij, 2021-02-28:
you can see all in this video - https://youtu.be/EhfY6-gJujc
After restart all fine.

UPDATE
After open chrome (or any other program) on full screen margin appears.

luckman212, 2021-02-28:
Oops- you must have posted this as I was writing #369 (sorry)

exelban, 2021-02-28:
It must help: `defaults delete eu.exelban.Stats`.
I must reset the settings in the new release. But was not sure how many users will have this bug.

n-osennij, 2021-02-28:
> It must help: `defaults delete eu.exelban.Stats`.
> I must reset the settings in the new release. But was not sure how many users will have this bug.

no. this command reset my settings, but even after restart issue is not solved.

exelban, 2021-02-28:
Hmm, thanks for the feedback. I will try to debug why it heppens.

exelban, 2021-02-28:
Could you maybe drag these empty spaces as in #369?
Or it just space?

n-osennij, 2021-02-28:
> Could you maybe drag these empty spaces as in #369?
> Or it just space?
[image]

just space. no ability to move it, click or doing something else

exelban, 2021-02-28:
Could you please check this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/6057842/Stats.dmg.zip)

n-osennij, 2021-02-28:
> Could you please check this build?
> [Stats.dmg.zip](https://github.com/exelban/stats/files/6057842/Stats.dmg.zip)

1) fully close Stats
2) open Stats.dmg you send for my
3) drag and drop to Programs folder, click Replace
4) run Stats, click Open in dialog
5) open chrome full screen
6) issue is not solved

n-osennij, 2021-02-28:
now try to delete Stats first. The same result - issue is not solved

exelban, 2021-02-28:
Thanks. Don't need to move it to the application folder. You can just run it from the .dmg.

What about this one?
[Stats.dmg.zip](https://github.com/exelban/stats/files/6057869/Stats.dmg.zip)

n-osennij, 2021-02-28:
> Thanks. Don't need to move it to the application folder. You can just run it from the .dmg.
> 
> What about this one?
> [Stats.dmg.zip](https://github.com/exelban/stats/files/6057869/Stats.dmg.zip)

it works! all is fine

exelban, 2021-02-28:
Nice. Thanks for the feedback. I will wait for confirmation from @luckman212 and make a new release after. 
@luckman212 please check if the last build fixes the issue for you.

luckman212, 2021-02-28:
Testing now @exelban ...

luckman212, 2021-02-28:
Looking good! Fixed it for me as well 🎉

exelban, 2021-02-28:
@n-osennij @luckman212 
Thank you for the feedback and fast replies)
I'm going to make a new release today with this fix.

richardtop, 2021-02-28:
Issue confirmed

exelban, 2021-02-28:
[Release](https://github.com/exelban/stats/releases/tag/v2.5.3) with this fix available.

closed 2021-02-28 by exelban

luckman212, 2021-02-28:
Fastest fix on a Sunday I've ever seen! Thank you 🦄