exelban/stats #371: Feature Request : Support for adding multiple 'disk read/write indicator'

acotor, 2021-03-01:
I hope to add more 'disk read/write indicator' for every single external usb hard drive of mine.

For now I can only add one indicator to show the read/write speed for one specific disk. I haven't found an alternative can do the same job. I don't want to benchmark my disk. I just want to monitor the transfer speed of a disk when copying files.

exelban, 2021-03-04:
Hi. You can see read/write per disk in the popup view.

acotor, 2021-03-05:
> Hi. You can see read/write per disk in the popup view.

Yeah, I can see those little blue/red squares indicators for each disk. But they only tell read/write status but no data speed(MB/s or etc.). Is there a way to customize the popup view content?

Let me clarify this. There is a menubar indicator can show disk speed already, which can be customized and can be seen without a click. Such features are similar to a gadget. I think both are good solutions. But for now, users can add only one — one for total disks or one for a specific disk.

exelban, 2021-03-07:
Hi. Unfortunately, it's not possible, for now.
It could be possible only when you could enable a few instances of the Disk module. Maybe it will be added in the future. But I cannot promise. There were similar asks for a few GPU monitoring (#93), a few network interfaces (#355), etc.

closed 2021-03-07 by exelban