exelban/stats #376: Feature request: Making vertical text next to icons toggle-able

molevision, 2021-03-08:
It would be beneficial to customisation and modularity to make each of the static vertical text elements (CPU, GPU, RAM) customisable, so that they can be turned on and off or rearranged like any of the others.

I'd also appreciate them being added for Network (NET), Sensors (SEN?) and Fans (FAN).

exelban, 2021-03-12:
Hi. Thanks for the proposal.
I think it will be a good option to remove the labels from some widgets and create a new widget type `Label`. And by enabling it users could use it as they will want.

There was an idea to have a label only if the same widget could be used in the several modules. But now (after adding the option to have a few widgets enabled) it will be better to create a separate 'widget' for this.

closed 2021-04-04 by exelban

molevision, 2021-04-04:
Great to see this implemented!

One small bug: The CPU graph box still has the "Label" option toggle-able, allowing for a duplicate.

A few further ideas for refinement: restricting the movement of the label element to only be placed adjacent to a parent widget when command-dragged, making the highlight area that is created when clicking it cover both it and the parent widget, and moving the label with the parent when the parent is command-dragged.

I'm unsure if the menu bar code for macOS allows these things, but they would make rearranging widgets lot more intuitive.

exelban, 2021-04-04:
Hi.

Yes, I know that the `Mini` widget has a label option. But it's not a bug, it's a feature)
I'm using a few of these widgets with labels. And it's important to me to have label + value in the same place. 
I have removed the `label` option only from widgets, which have the same label as in the new widget. But as you can notice, the `Mini` widget has a little bit different label (on the top, not on the side).

Thanks for the idea. But I'm not sure it's possible. First, it's could be not makable from the 'code' point. Second, I don't think I will ever find a time for a feature like this.

BenTalagan, 2021-04-05:
Thank you for not having removed the "label" option. On Big Sur, all menu bar widgets have an extra side space, which makes the single standalone label widget a no-go for me (to my humble personal taste, of course, no offense 😃 ).