exelban/stats #379: Feature request:  toggle CPU frequency

tlove004, 2021-03-09:
Intel Power Gadget is fairly resource intensive; it would be nice to be able to toggle this on/off in the CPU widget.

exelban, 2021-03-09:
Hi. Yes, it could be added.
But I'm not sure if it makes sense. The app reads data from IPG only when the popup is open. Yes, it will initialize IPG on start, but only initialize.

tlove004, 2021-03-09:
Perhaps the toggle could have a prompt/warning "The application will need to restart."  Then, IPG will not be loaded on start if it is toggled off.

exelban, 2021-03-09:
Did IPG in your case use a lot of CPU?

tlove004, 2021-03-09:
Actually, it was a memory issue; IIRC, when profiling in Instruments I noticed a memory leak related to loading IPG.

exelban, 2021-03-09:
yeap, there are some leaks in the memory. But I'm not sure if it's related to the IPG. I suppose it's related to the linked library + allocations in the IPG (trying to guess). Because the are some leaks (~100) only when initializing the IPG.

tlove004, 2021-03-09:
I only profiled for short periods of time, so I'm unsure if it would have stabilized at some point, but it did seem to be increasing with time.  In any case, disabling loading IPG altogether was a happy compromise for me, since I don't care too much about CPU frequency on a regular basis, and there have since been no memory issues that caught my attention.

exelban, 2021-03-09:
Ok, I think I could add this option.

closed 2021-03-14 by exelban