exelban/stats #380: Feature request: Option to set chart width

chmac, 2021-03-12:
Firstly, thanks for this package, it's really excellent, I'm delighted to have found a new update for menumeters. Thanks.

One feature I miss from menumeters is the option to change how wide the chart widgets are. When I'm on the big screen, I'm happy to double or triple the width of the charts so I can see more of my CPU load history. I hope I explained that clearly, if not, just let me know and I'll post an image of what I mean.

exelban, 2021-03-12:
Hi. Thanks for the feedback.
Yes, I think it's possible. I think you propose to change the time range which is visible in the chart widget. Proportionally changing the width, depends on the time range?

chmac, 2021-03-12:
Yes, that's exactly what I'm thinking! Is it already possible in the app? I tried looking in settings but I couldn't figure out how to do it.

exelban, 2021-03-12:
Nope. There is no option to do it. But it could be added.

chmac, 2021-03-12:
@exelban Awesome, thanks for the fast response. Alas I'm not familiar with swift, so I can't offer much help on the coding front, but happy to test beta releases / etc if that's any use.

closed 2021-12-05 by exelban