exelban/stats #381: The order in menu bar cannot be stored after reboot of machine.

Tanche-Z, 2021-03-14:
**Describe the bug**
Hi~ I find the order in the menu bar can't be stored after my customize when restarting the system.
I'm wondering if we can add a feature to customize the order shown in the menu bar.

**Details:**
 - Device: MacBook Pro 13.3 2020
 - MacOS 11.2.2
 - Application version 2.5.4

exelban, 2021-03-14:
Hi.
Does it always starts with the same order? Or each time with random?

Tanche-Z, 2021-03-15:
In most of time, the widgets load in same order. If I only activate network, GPU, RAM, CPU, it will load in that order every time. Wondering if the default order can be customized and stored?

exelban, 2021-03-15:
Yes, it's already saving the position. #358 #369

Tanche-Z, 2021-03-16:
Thanks for reply!

The release version is 2.5.5.

If I set the order like this (by activate them one by one):
[image]

Then, after reboot it will be like this:
[image]

It does have a fix position, but what if it can be customized?:sparkles:

exelban, 2021-03-16:
And what if you will order the items manually?

closed 2021-03-16 by Tanche-Z