exelban/stats #383: Network Widget - Cannot copy IP address via Cmd+C keyboard shortcut

JeffreyCA, 2021-03-17:
**Describe the bug**
A clear and concise description of what the bug is.

**Details:**
 - Device: Macbook Pro 2019
 - MacOS 11.2.3
 - Application version: 2.5.5

In the network widget, I can double click the IP address fields to select the addresses, but pressing Cmd+C does not copy it to system clipboard. I have to right click + copy. And it seems that click to copy functionality was removed?

exelban, 2021-03-17:
Hi. Yes, `click to copy` was temprorary removed. Hmm, very strange that it cannot be copied to CMD+C. I will take a look at this. 
And I think `click to copy` must be returned.

JeffreyCA, 2021-03-26:
Thanks, it's fixed now.

closed 2021-03-26 by JeffreyCA

exelban, 2021-03-26:
Yeap. I was leaving this ticket as a reminder to make the `click to copy` feature.