exelban/stats #384: file descriptors leak

pinpins, 2021-03-19:
```
 sudo lsof -n | cut -f1 -d' ' | uniq -c | sort | tail
Password:
 306 Microsoft
 312 Code\x20-
 334 Google
 351 cloudd
 381 Google
 389 MySQLWork
 426 corespotl
 641 Code\x20-
 676 Microsoft
4914 Stats
```

5000 too many fds on mac

exelban, 2021-03-19:
Hi. Could you please explain?

pinpins, 2021-03-19:
Stats program has used 4914 file descriptiors, far too much, and macos was erroring with "too many opened files" error.

Seems your program opens too many files(proceses as file is process) and don't close them properly. 

See, why it would require it to have 4914 open fds.

exelban, 2021-03-19:
I suppose you have opened Stats for a long time because I don't see such descriptors. 
But yes, it looks like I must take a look if I close all processes properly. Does it cause bugs?

pinpins, 2021-03-19:
yes, macos starting acting up.

exelban, 2021-03-19:
For how long you have been using the app?
I mean after what period of time (+-) you have such opened fds?

pinpins, 2021-03-21:
like 24h? maybe 40

exelban, 2021-03-21:
Hmm, I have been running the app for a few days. And does not have this issue.
Looks like some module cause this. Which I don't use. If you will have an idea of what exactly causes this, give me to know.

exelban, 2021-03-21:
Maybe you keep opened the popup window?

closed 2021-03-28 by exelban