exelban/stats #386: Other (Plug-)Battery-Icon when battery isn't charging

oeppfel, 2021-03-23:
It would be nice if there was (similar to macOS) a plug-like icon in the battery-icon when the battery isn't charging. I know that there is the possibility to hide additional information, but only when the battery is fully charged, not when charging is inhibited from a specific level by some application.

exelban, 2021-03-23:
Hi. I'm not sure if I understand correctly. Could you please explain what do you propose?

oeppfel, 2021-03-23:
Hi :)
I replaced the native battery status bar icon with yours from stats and saw that, different from macOS, the status bar icon doesn't change when the battery isn't charging. I propose that the icon shows (like in the native battery icon) a plug-like symbol instead of a flash to signalise that the battery isn't charging and the computer is powered with AC power.
Kind regards

exelban, 2021-03-23:
So, you propose to visualize in some way that battery is charging. Because it's hard to say if the battery is already charged for 100% or only for 99%. Because the flash is always present (when charging and when fully charged)?
Do I understand correctly?

exelban, 2021-03-23:
I propose you enable the `Colorizing` function. It will colorize the inner when the battery not fully charged. And monochrome when fully charged.

[image]
[image]

closed 2021-04-04 by exelban