exelban/stats #391: Battery Dashboard doesn't show processes

beedaan, 2021-03-29:
**Describe the bug**
The battery dashboard can show top processes.  I've got this option set to 8.  I upgraded my Stats version this morning and noticed it isn't working.

[image]

**Details:**
 - Device: MacBook Pro 16-inch 2019
 - MacOS: 11.2.3
 - Application version: 2.5.7

ppizarror, 2021-03-30:
I also can confirm this issue.

- Device: MacBook Air M1, 2020
- MacOS: 11.2.3
- Application version: 2.5.7
[image]

exelban, 2021-03-30:
Hi. Thanks for reporting.
I have noticed this bug yesterday. And there is already [fix](https://github.com/exelban/stats/commit/b802bd6b40a357f3be37f2e78cdb4a4e94083809).

closed 2021-04-04 by exelban