exelban/stats #394: No information about memory / graphics in Dashboard - M1

ppizarror, 2021-03-31:
The current dashboard page does not show information about Memory and Graphics.
I'll take a look at this issue.

- Version: 2.5.7

[image]

exelban, 2021-03-31:
Hi. Thanks for reporting. I know about this. But haven't the option to fix it.

For RAM, could you please share the output of `/usr/sbin/system_profiler SPMemoryDataType -json`?

ppizarror, 2021-03-31:
Sure, 

```bash
└─[$]> /usr/sbin/system_profiler SPMemoryDataType -json
{
  "SPMemoryDataType" : [
    {
      "dimm_type" : "lp_ddr4",
      "SPMemoryDataType" : "8 GB"
    }
  ]
}%
```

As I see in SystemKit.swift, ```public func getRamInfo() -> ram_s? {``` assumes that the json has some structure not present in that -json. Also, GPU properties can be gathered using

```bash
└─[$]> /usr/sbin/system_profiler SPDisplaysDataType -json
{
  "SPDisplaysDataType" : [
    {
      "_name" : "kHW_AppleM1Item",
      "spdisplays_metalfamily" : "spdisplays_mtlgpufamilyapple7",
      "spdisplays_ndrvs" : [
        {
          "_name" : "Color LCD",
          "_spdisplays_display-product-id" : "a03c",
          "_spdisplays_display-serial-number" : "...",
          "_spdisplays_display-vendor-id" : "610",
          "_spdisplays_display-week" : "0",
          "_spdisplays_display-year" : "0",
          "_spdisplays_displayID" : "1",
          "_spdisplays_pixels" : "2880 x 1800",
          "_spdisplays_resolution" : "1440 x 900 @ 60.00Hz",
          "spdisplays_ambient_brightness" : "spdisplays_no",
          "spdisplays_connection_type" : "spdisplays_internal",
          "spdisplays_display_type" : "spdisplays_built-in_retinaLCD",
          "spdisplays_main" : "spdisplays_yes",
          "spdisplays_mirror" : "spdisplays_off",
          "spdisplays_online" : "spdisplays_yes",
          "spdisplays_pixelresolution" : "spdisplays_2560x1600Retina"
        },
        {
          "_name" : "SyncMaster",
          "_spdisplays_display-product-id" : "55f",
          "_spdisplays_display-vendor-id" : "4c2d",
          "_spdisplays_display-week" : "8",
          "_spdisplays_display-year" : "2009",
          "_spdisplays_displayID" : "2",
          "_spdisplays_pixels" : "1920 x 1080",
          "_spdisplays_projector" : "spdisplays_yes",
          "_spdisplays_resolution" : "1920 x 1080 @ 60.00Hz",
          "spdisplays_ambient_brightness" : "spdisplays_yes",
          "spdisplays_mirror" : "spdisplays_off",
          "spdisplays_online" : "spdisplays_yes",
          "spdisplays_pixelresolution" : "spdisplays_1080p",
          "spdisplays_resolution" : "1920 x 1080 @ 60.00Hz",
          "spdisplays_rotation" : "spdisplays_supported"
        }
      ],
      "spdisplays_vendor" : "sppci_vendor_Apple",
      "sppci_bus" : "spdisplays_builtin",
      "sppci_cores" : "7",
      "sppci_device_type" : "spdisplays_gpu",
      "sppci_model" : "Apple M1"
    }
  ]
```

I inspected the whole output from ``` /usr/sbin/system_profiler -detailLevel mini```but that's the only information regarding memory and GPU

closed 2021-04-04 by exelban