exelban/stats #396: extFS appears as full in Disk menu

thebitstick, 2021-04-01:
**Describe the bug**
extFS Volumes mounted by extFS for Mac by Paragon Software appear as 100% full under the Disk monitor.
Disk Utility, Finder, and df (Terminal.app) are all able to understand and measure disk usage for extFS Volumes.

**Details:**
 - Device: Mac mini 2020
 - MacOS 11.2.3
 - Application version 2.5.7

exelban, 2021-04-01:
Hi. Thanks for the feedback. I can't ensure that disk space will be correct when you using external drivers for it.

closed 2021-04-04 by exelban

thebitstick, 2021-04-10:
This has somehow been fixed??
[image]
None of the commits would suggest that.
Perhaps a bug with the external driver.