exelban/stats #405: No battery level notifications

chetanpm, 2021-04-05:
**Describe the bug**
In the battery section, I've selected the low level notification to 30% and the high level notification to 80% but I don't get any notification when the battery levels go past these values.
 
**Details:**
 - Device:  Macbook Pro 2013
 - MacOS: 11.2.3
 - Application version: 2.5.7

exelban, 2021-04-05:
Hi. Thanks for reporting. I will check it.

exelban, 2021-04-06:
Hi. I check low and high level notifications. Both works as expect.
Notice please that low level will be only when not charging. And high level - only when charging.

chetanpm, 2021-04-07:
[image]
[image]

https://user-images.githubusercontent.com/9249405/113811507-ee582580-9789-11eb-8010-b3c050038439.mov

What am I doing wrong?

exelban, 2021-04-07:
Hm, it's really strange. Could you maybe check if you have no notifications in the notification center from `Stats`?
I suppose low level also does not work?

chetanpm, 2021-04-07:
The first time stats was run, it asked do you want to allow the application to send notifications and I selected 'allow'. I have never received any notification from stats ever. It doesn't notify for low battery level either.

exelban, 2021-04-07:
There are only a few options when notification will not be delivered:
- notification already present in the notification center
- macos does not deliver notification
- application is currently active and then the user receives a "silent" notification. It means that notification will just appear in the notification center

chetanpm, 2021-04-07:
Ok, I use fan speed control. So if the app is running with 'root', the notification doesn't work. Restarting the application with no manual fan control, I get the notifications properly.

I wanted the application because it replaced macsfancontrol and iglance and gave me a way to control fan and notify battery charge level in one so I always ran the application with 'root' rights and thats why I thought the notification didn't work. 

Can you check the notification with root access and see whats the problem?

exelban, 2021-04-07:
Yeap, I will try to fix it.

exelban, 2021-04-11:
Hi. I check this bug. And it looks a little bit unreal to fix. 

TL.DR. Notification will be delivered to the user who has run the app. So, when the app runs under root, a notification receives root.

The problem: to send the notification I must use `NSUserNotificationCenter`. There is only one way to make a notification. And it looks like this module is connected with user "space". So the notification will be sent to the user who runs the app. In a normal use case, the app is running by an active user. And it user receiving the notification. But when you running the app as root. The notification also is delivering to the root. 
I'm not sure if it's fixable. Because it looks like good behavior.

closed 2021-04-11 by exelban

chetanpm, 2021-05-26:
Can't you use 'sudo' to give root access to the current user. By editing sudoers files, one can grant admin access to their user account. It is much more flexible that way and of course, the notifications will come to the user running the process.

exelban, 2021-05-31:
Hi. I think it will be fixed when #454 will be closed. After that, the Stats will not be running under root.

exelban, 2021-06-25:
This issue must be resolved since #454 is released. Because the app does not run under root.