exelban/stats #409: Battery impact on Apple Silicon

cristiansori2, 2021-04-06:
Im having a very high battery impact on MacBook Pro M1, 4.8% each hour according to Activity Monitor.Is this ok?

**Details:**
 - Device: [MacBook Pro M1]
 - MacOS [11.1]
 - Application version [2.5.6]

exelban, 2021-04-06:
Hi. Thanks for the feedback. Could you please share the statistics from Activity Monitor?

Jethro1000, 2021-04-07:
I can't confirm that stats is using the battery excessively. Of course, if you set all metrics to 1 second, this results in higher consumption.
My settings:
CPU / GPU / sensor = 2 sec
Disk = 10 sec
Network = off
Fan = should at 5 or 10 sec, but it didn't work on my MBP M1

exelban, 2021-04-07:
I'm trying to keep power consumption on an acceptable level. Of course, it cannot be around 0. Because there is a lot of readers, which collect the data about CPU, RAM, GPU, etc. But almost 60-70% of CPU time takes rendering widgets in the menu bar. And I suppose power consumption depends more on the number of widgets you have enabled.

closed 2021-04-09 by exelban