exelban/stats #413: Fans missing

asheenlevrai, 2021-04-09:
Hi :)

I just found out about stats. Great piece of software! Congrats!!
I am currently running it on a Ryzentosh running Catalina. (let me know if HW specs would be relevant here).

It appears, in my hands, there are 0 info/control related to fans while the Features list for stats includes "fan speed" and "fan control".

Is there anything I could do to change that? I mean to have the info/controls over my fans through stats?

Thanks a lot in advance for your help.

Best,
-a-

exelban, 2021-04-09:
Hi. Yes, Stats has a `Fans` module that allows you to see fans speed and control it.
I suppose `Ryzentosh` is some kind of Hackintosh. So you need to ensure if fans is available in your system.

asheenlevrai, 2021-04-09:
Thanks :)

Yes, a Ryzentosh is a hackintosh based on an AMD Ryzen CPU.

I don't quite understand what you mean by "_you need to ensure if fans are available in your system_" (excuse mi ignorance).

exelban, 2021-04-09:
You need to have fans available throw `SMC`.

asheenlevrai, 2021-04-09:
You're right.

I tried a different app on this Ryzentosh and here too the fans were not detected.
I also tried stats on a different hackintosh and in this case the fans are properly detected.

I guess I need to look into how to have the fans available through SMC (I don't even know what that means right now tbh).

Thanks a lot :)
-a-

exelban, 2021-04-09:
You need to take a look at the SMC which you use. FakeSMC or VirtualSMC could not properly detect fans from the motherboard. Sometimes just enabling some keys helps. Sometimes only ACPI.

closed 2021-04-09 by exelban

asheenlevrai, 2021-04-12:
https://dortania.github.io/OpenCore-Install-Guide/troubleshooting/extended/post-issues.html#no-temperature-fan-sensor-output

 found this