exelban/stats #414: [Apple M1] Fans not displayed

Jethro1000, 2021-04-11:
**Describe the bug**
Fans Section not displayed

As you can see in picture, i set two breakpoints in Modules/Fans/main.switft (line 71 and 75) and receiving following error:
2021-04-11 18:52:43.778479+0200 Stats[54742:6577158] [Array<Fan>] Successfully initialize reader
Error call(READ_KEYINFO): (iokit/common) invalid argument
Error read(): (iokit/common) invalid argument

[image]

**Details:**
 - Device: Macbook Pro M1 2020
 - MacOS: 11.2.3
 - Application version: 2.5.9

Jethro1000, 2021-04-11:
If you need access to a M1 MBP, i can arrange this. Please let me know

exelban, 2021-04-11:
Hi. Thanks for the reporting and sharing the logs.
It's a very interesting problem. I will try to debug why it's throwing the error.

exelban, 2021-04-11:
Do you maybe any open source software showing fans on apple silicon (natively)?

exelban, 2021-04-11:
Because as I know apple silicon does not have SMC at all.

exelban, 2021-04-23:
I need an m1 to ensure the proper work of sensors and fans module.
I will try to find one, but it will take some time.

altrdev, 2021-05-04:
Installed this:
https://github.com/iglance/iGlance

RPM are visible in Apple Silicon machine

Jethro1000, 2021-05-04:
Correct, but iGlance is running under Rosetta2. It's a x86_64 Code not Universal

AchinthyaBhat, 2021-05-26:
Has anyone tried to asking for help at apple.stackexchange.com or any similar forums? I would, but I wouldn't know what to do with the answers 😅

exelban, 2021-05-27:
Hi. Not me.
I'm checking different forums from time to time. But haven't seen anything related to this issue.

exelban, 2021-06-10:
I'm not sure if I understand you correctly.
They have their own SMC kext or what?

bassu, 2021-06-22:
@exelban
It's not using a special kext (I found none in `kextstat`).

However, I did end up finding the exact SMC keys you can query (had to reverse engineer a couple of apps and I used to the tools name list `nm` and `objdump` 😀  ).

Use any SMC library 📦 out there to your liking and query these keys. There's one in Swift by [beltex](https://github.com/beltex/SMCKit) 

Notes: 
- My 13" M1 has just one fan so the key is `F0Ac` for current speed and `F0Tg` for target speed.
- Increment it for more fans i.e. `F1Ac`, etc.
- If querying the key `F0Ac` does not return any value, then that fan is assumed _off_

Below is the live data of my machine I was able to query using SMC, without using any x86 code.
I had to run CineBench for 5 mins to kick in the fans 😜 

[image]

References:
- https://web.archive.org/web/20150107044816/http://www.parhelia.ch/blog/statics/k3_keys.html
- https://opensource.apple.com/source/net_snmp/net_snmp-171/SensorDat.xml

Enjoy!
🕊️

bassu, 2021-06-25:
I see it's already being done here in _[readers.swift](https://github.com/exelban/stats/blob/13707090eb39f29024a93f525c2cd7822329368b/Modules/Fans/readers.swift#L33)_ so perhaps this issue has something to do with the way the Stats' SMC library is fetching these values. 

Because I used the Golang bindings for the SMC and they worked flawlessly to fetch all fan parameters above ☝️

exelban, 2021-06-25:
@bassu Hi. Could you confirm that there is an SMC on macs with apple silicon?
Because all that I was found on the internet says that there is not SMC.

Jethro1000, 2021-06-26:
In fact, i can confirm that M1 Macs did **not** have a SMC. All functions that were previously handled by the SMC are now embedded in the M1 processor

bassu, 2021-07-03:
@exelban @Jethro1000 
M1s do have an equivalent of an SMC.

I can read about 1435 keys from the SMC. Even the connection string name is the same i.e. `AppleSMC`.

For instance, the original [smcFanControl](https://github.com/hholtmann/smcFanControl/blob/master/smc-command/smc.c) used to ship C tool called `smc` that still works after you compile it and gives you correct readings.

[image]

mrchoke, 2021-07-22:
@bassu  great work. I test on mac mini M1. it can show rpm in system bar too!
 
[image]

sdkdimon, 2021-11-07:
Made pull request for this issue.
https://github.com/exelban/stats/pull/689

exelban, 2021-11-07:
Hi. Please, someone who has apple silicon, check if fans are available in the new [build](https://github.com/exelban/stats/releases/tag/v2.6.18).

Jethro1000, 2021-11-07:
Confirmed, it's working now. Thx

Jethro1000, 2021-11-07:
[image]

closed 2021-11-07 by Jethro1000

exelban, 2021-11-07:
thanks to @sdkdimon