exelban/stats #415: Feature Request: Add average temperature to selected sensors

raffaelepojer, 2021-04-11:
It would be great, if it is possible, to average the temperature for the selected sensors (Mac M1).
Thanks!

exelban, 2021-04-11:
Hi. Thanks for the proposal. What do you mean by `average temperature to selected sensors`?
I think I can add a small chart to each sensor. And it could show some `average` temperature.

raffaelepojer, 2021-04-11:
I mean to calculate the average of the values of the selected sensors instead of seeing some specific temperature only

exelban, 2021-04-11:
Could you maybe propose some interface for this feature?

raffaelepojer, 2021-04-12:
It can be the same as it is now with the only difference is that a new toggle computes the average of the selected sensors (temperature)

raffaelepojer, 2021-04-12:
I have had a look at other monitor applications and for example, TG pro seems to take the highest temperature of the Performance Cores

exelban, 2021-04-18:
Hi. I'm not sure that I could implement this feature in a good way. It requires too complicated UI for such a small feature.

closed 2021-04-18 by exelban