exelban/stats #416: Label removal on pie charts / graph widgets forces to have extra spaces on Big Sur

BenTalagan, 2021-04-11:
Hi, feel free to close this issue if you don't think it's relevant regarding the direction you want to impulse to `stats` 😃 

Still, I thought it might have been worth noting that on Big Sur, having labels as independent widgets forces us to have an extra space between the independent label widget and any another widget. My humble preference goes for the older feature where widgets would optionally embed their own label thus allowing for a more "compact" layout, more readable. On Big Sur a lot of space is already wasted in the menu bar due to that extra space. So I'm not really a fan of commit 3ac95f99523f7806ae31ef1aad574b039a7195fd !

The precedent version of stats (2.5.8) was really cool because both options were permitted (embedded label or independent label).

Thanks for the great work on `stats` anyway !

exelban, 2021-04-11:
Hi. Thanks for the feedback.
Hmm, I see. I think I can revert these changes. And return the label option for some types of widgets.

BenTalagan, 2021-04-11:
Thanks a lot for your lightning-fast answer ! 😃

ppizarror, 2021-04-12:
HI, I had the same question regarding this issue.  Maybe an optional toggle switch for "show label" may be a good addition

closed 2021-04-18 by exelban