exelban/stats #417: Memory leak on 2.5.9

PriZmRL, 2021-04-11:
There's some massive memory leak in 2.5.9, the memory usage climbed from 50 mb to 438 mb in just an hour, see the screenshot

The Features i've turned on are;

- CPU
- GPU
- RAM
- SENSORS
- NETWORK

**Details:**
 - Device: M1 Macbook Air
 - MacOS 11.2.3
 - Application version: 2.5.9
 - 
[image]

exelban, 2021-04-12:
Hi. Thanks for reporting.
Could I ask you to disable the sensors module and check if the leak still present?

PriZmRL, 2021-04-12:
Disabling sensors module indeed fixes it. Ram usage does not increase anymore, but it's not decreasing until i restart Stats either

exelban, 2021-04-12:
Yes. It's normal. Thanks, now I know that it's a leak in the m1 sensors reader. I will fix it.

exelban, 2021-04-12:
I make a build with a fix. Please check if the leak disappears.
[Stats.dmg.zip](https://github.com/exelban/stats/files/6299268/Stats.dmg.zip)

PriZmRL, 2021-04-12:
installing right now. Will let you know!

PriZmRL, 2021-04-12:
so far i see the ram usage is increasing about 1.5 mb's per minute, when i started the app it had 45 mb ram usage, now its at 57 mb and rising slowly

exelban, 2021-04-12:
Small raise with time it's ok. Opening a popup view or settings will allocate some memory. 
Please keep the app working for some period of time.

PriZmRL, 2021-04-12:
Sure thing. I'll keep it on and will report back in about an hour. Currently it's at 65 mb usage, randomly checking modules on menubar at times

PriZmRL, 2021-04-12:
So about 1 hour passed, ram usage is at 125 mb and climbing. Without sensors module in previous versions it would stay at about 55-60 mb during the day

exelban, 2021-04-12:
Thanks, I will keep looking.

exelban, 2021-04-12:
Check please this one.

[Stats.dmg.zip](https://github.com/exelban/stats/files/6300054/Stats.dmg.zip)

PriZmRL, 2021-04-13:
Sadly no effect. It's at 250 mb usage right now

exelban, 2021-04-13:
(
Unfortunately, I could not reproduce this bug because it presents only on apple silicon.
Only someone with m1 and xcode could help to find and fix this leak.

PriZmRL, 2021-04-13:
Most of the voltages and currents showing value 0, maybe its because of that? there's also 2 sensors that are showing negative value of -22c all the time, could be causing issue as well

exelban, 2021-04-19:
Hi. Could I ask you to check this build?
It will show always `1` as value for all sensors. It's ok. I just trying to figure out where is the leak.

[Stats.dmg.zip](https://github.com/exelban/stats/files/6337067/Stats.dmg.zip)

PriZmRL, 2021-04-19:
Sure thing

RajceP, 2021-04-19:
On it.

PriZmRL, 2021-04-19:
Ram usage seems to stay at 72.3 mb with sensors module enabled.

*CPU
*GPU
*RAM
*NETWORK
*BATTERY
*SENSORS

I currently have these modules enabled. I think sensors module is heavy on the resources. When i have rest every module except sensors enabled, the cpu usage is max %1 - 1.5% and 45 mb with all modules enabled. 

With sensors module, cpu usage goes to 4% and stays at 72.3 mb.

RajceP, 2021-04-19:
RAM stays at 59,1 mb for me.

colebob9, 2021-04-28:
Ran the latest commit through the Instruments profiler on my M1 Mac Mini with the sensor widget on. Simply left it open for a few minutes without touching anything so the results are consistent. The profiler is reporting leaks constantly and it is gradually growing in memory usage as time goes on like others have reported. 

Something with the IOKit library is causing the leaks as suggested from the screenshot below.

[image]

Running the same test without the sensors widget seems to nail down that this is definitely the sensors widget causing the issue. No leaks reported after that first one shown, and the memory usage stays consistent under 25 MiB.

[image]

exelban, 2021-04-28:
@colebob9 Hi. Thanks for debugging. I have exactly what function causing the leak. But I cannot reproduce this leak on intel to fix it(

exelban, 2021-05-06:
Hi. Could someone check this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/6437551/Stats.dmg.zip)

RajceP, 2021-05-06:
> Hi. Could someone check this build?
> [Stats.dmg.zip](https://github.com/exelban/stats/files/6437551/Stats.dmg.zip)

Looks promising! I am under 50mb after 20 minutes. I will report tomorrow.

RajceP, 2021-05-07:
62.2mb after 8 hours :)

exelban, 2021-05-07:
Thanks for the feedback.

closed 2021-05-09 by exelban