exelban/stats #420: Mac Mini M1 - Network traffic numbers are very small and barely visible

phattrance, 2021-04-12:
**Describe the bug**

The stats for network traffic and HDD write / read gets placed above each other, which makes them very small and hard to read on bigger screens (Specially compared with the other numbers like CPU / GPU etc). Can we get a number to place them next to each other instead of above each other? That should solve the problem with small numbers.

Screenshot: https://i.imgur.com/Mi6e3rN.png

**Details:**
 - Mac Mini M1
 - Big sur 11.2.3
 - Application version 2.5.9

exelban, 2021-04-12:
Hi. Thanks for the feedback.

I'm not sure that it will be a good feature. The `mini` widget (CPU/RAM etc) is good visible because it must show only one value. And this value is in the range of 0-100. And it's each to fit in some fixed width. There is only one option to make the `speed` widget bigger. It's to put values next to each other (as you propose). But then will be a little problem. Or the widget of the widget will be too big but static. Or width will be dynamic, but you will see a "jumping" menu bar. I would prefer the fixed. Because the speed is highly dynamic. And I don't like the "jumping" menu bar.

Example: 100% vs 128 KB/s | 128 KB/s. The first one it's only 4 symbols. The second one - 19. So the width of this widget will be too big.

I'm just interested, what size of your screen? And what the resolution?

exelban, 2021-04-12:
Additional question. What do you use for this line separator between widgets?
And for these big labels (RAM, GPU, HDD)?

phattrance, 2021-04-12:
Hi,

Im using a 49" Samsung G9 running on 5120x1440 @120hz on my Mac Mini (m1). The Extra added separators are from an menu bar called Bartender, Its a good match to have the bartender and stats app working togeather. 

Regarding the jumping menu bar, cant that be solved by having an option to choose displaying in MB/S instead of Mbit/s?  Or make so that when reaching over 1000kb/s, it will auto switch to 1 MB/S (or somethign similar)

 For us that have 1gbit conenction, we rather see the stats in MB/S that mbits :)

exelban, 2021-04-12:
It automatically switches between KB -> MB. When reaching more than 3 numbers. But still 3 numbers + suffix x2. It's a lot of symbols.

Wow, it's a really big monitor. In that case, you definitely don't see these small numbers(

phattrance, 2021-04-12:
Maybe you can solve this by adding an option to show only in MB/S (for those that want to only see this)? And this way can have have one line per D / U.

Yeah with this resolution / screen i need to be like 15cm from the monitor to see the network and disk stats. Its pretty much not readable when you are 30cm+ from the screen.

exelban, 2021-04-12:
No, I will not add a hack for this. In your case, you need only MB/s. But someone is more interested in KB/s. It will be a little bit hard to handle all cases. Also, provide unnecessary complications in config.

It requires 2 additional settings to cover this case: type of rows (vertical/horizontal) and option to show/hide suffix.
I'm not sure that I will find time to implement these options. Additionally, it's a super unusual use case.

phattrance, 2021-04-12:
ok, thx for the help!

exelban, 2021-04-12:
Please reopen the issue, if you will have a solution for this case.

closed 2021-04-12 by exelban