exelban/stats #421: User settable spacing between widgets

SidSidSid16, 2021-04-13:
A clear and concise description of what the bug is.

Not really a bug, just the fact that the default spacing between everything, including menu bar items, on macOS is excessively big. With more of these widgets placed on the menu bar, it starts to eat up a lot of space. Here's a screenshot for some context:

[image]

Would it be possible to add a feature that will allow users to change the spacing between widgets?

 - Device: MacBook Pro (15-inch, 2017)
 - MacOS 11.2.3 
 - Application version 2.5.9

exelban, 2021-04-13:
Hi. Thanks for your feedback.
On macOS Big Sur system add spacing to the elements in the menu bar. I remove all spaces on big sure. On the previous one there are some internal spaces, but not on Big Sur.
And unfortunately, Big Sur spaces could not be changes. There no option in the system to do it.

closed 2021-04-13 by exelban

SidSidSid16, 2021-04-13:
Ah, that makes sense, thank you very much.