exelban/stats #430: cant control Fans

CharlieJ0hn, 2021-04-19:
**Describe the bug**
A clear and concise description of what the bug is.

**Details:**
 - Device: MacBook Pro 16
 - MacOS 11.2.3
 - Application version 2.5.10
[image]

exelban, 2021-04-19:
Hi. Thanks for the feedback.

Could you please provide more details?
Do you use any other app to control the fans at the same time? 
What if you try to control only one fan?

CharlieJ0hn, 2021-04-22:
sorry,i have no more details ,so icant provide.
yes ,i user istat menus can controller the fans.
i got the same result  when i try to control only one fan, its not work.

CharlieJ0hn, 2021-04-22:
oh,i can tell you my macbook is 2.3 GHz 8cores Intel Core i9

exelban, 2021-04-23:
Hm, very strange. It could be some internal error in the SMC. Only debug build could give more information. Give me to know if you would like to run a debug version of app and share the logs.

ehajri, 2021-04-25:
@exelban, I am also not able to control the fans. I only use Stats, and briefly installed and uninstalled smcFanControl earlier this week.

System Specs:
Stats 2.5.10
macOS Big Sur 11.2.3
MacBook Pro (16-inch, 2019) Intel i9.

I am willing to run the debug version if you want.

CharlieJ0hn, 2021-04-26:
@exelban ok，i would like to run a debug version of app and share the logs.
and @ehajri would like too.
now we can share two log

exelban, 2021-04-26:
Hi. There is a debug build. It writes some logs to the `log.txt` which could be found in the `Documents`.
Please run this build and try to change the fan speed. And share the log.txt after. Thanks.

[Stats.dmg.zip](https://github.com/exelban/stats/files/6376834/Stats.dmg.zip)

CharlieJ0hn, 2021-04-26:
[image]
nothing in the log.txt but the time

CharlieJ0hn, 2021-04-26:
i think you can make a mode switch in the release vision to turn on/off the debug mode. 
everyone can send the log without reinstall the debug version when get an error.

exelban, 2021-04-26:
Thanks. Ok, I will take a look at another place.

Yeap, there was a special argument for this. But in 99% there was needed extra logs. So still every time I was needed to build a special version.

exelban, 2021-04-26:
Could you please try this one?

[Stats.dmg.zip](https://github.com/exelban/stats/files/6378984/Stats.dmg.zip)

ehajri, 2021-04-26:
> Could you please try this one?
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/6378984/Stats.dmg.zip)

log file is empty (except for date and time).

exelban, 2021-04-26:
Thanks. It's strange, what about this?
[Stats.dmg.zip](https://github.com/exelban/stats/files/6380382/Stats.dmg.zip)

ehajri, 2021-04-26:
> Thanks. It's strange, what about this?
> [Stats.dmg.zip](https://github.com/exelban/stats/files/6380382/Stats.dmg.zip)

Same thing, no logs just date/time header.

CharlieJ0hn, 2021-04-27:
i have the same content with @ehajri

exelban, 2021-04-27:
Oh. I have found the problem. For controlling the fans app needs to be run as root. And logs will be written to the root Documents too.
Please take a look for `/var/root/Documents/log.txt`.

ehajri, 2021-04-28:
@exelban since I had to do `sudo mkdir /var/root/Documents` first.

> ------------ 2021-04-28 01:46:37 +0000 ------------
current mode = 0, new mode = 1
change fan speed
change fan speed to 2148.8792161207934
change fan speed
change fan speed to 2152.870351938101
change fan speed
change fan speed to 2152.870351938101
change fan speed
change fan speed to 2152.870351938101
change fan speed
change fan speed to 2152.870351938101
change fan speed
change fan speed to 2152.870351938101
change fan speed
change fan speed to 2152.870351938101
change fan speed
change fan speed to 2152.870351938101
change fan speed
change fan speed to 2152.870351938101
change fan speed
change fan speed to 2152.870351938101
change fan speed
change fan speed to 2152.870351938101
send command change fan speed to the SMC
min = 1836, max = 5616, current = 1838, set = 2152

exelban, 2021-04-28:
@ehajri thanks. This is all output?

ehajri, 2021-04-28:
> @ehajri thanks. This is all output?

Yes, that is all.

exelban, 2021-04-29:
Thanks. Strange, so it looks like there are no errors.
So it can be a system, but to be sure need to check with SMC. If you would like to help with debugging give me know.
You will need to run one opensource tool which allows reading SMC value.

exelban, 2021-04-30:
This one: [link](https://github.com/hholtmann/smcFanControl/tree/master/smc-command)

CharlieJ0hn, 2021-05-01:
i see the project warning.
my mac is out of warranty period.
i dont know how to repair my macbook by myself.
so can you try it ? are you in apple care plus time?
@ehajri

exelban, 2021-05-02:
Yes. There is a warning. But it for writing keys to SMC. You can safely read from SMC (a lot of monitoring apps using it).
`./smc -l | grep F` - this command will return all key/value related to the fans.

CharlieJ0hn, 2021-05-03:
ok，i will try it to get the fans value

exelban, 2021-05-03:
Run this command when some fan will be in automatic mode, and in manual.

exelban, 2021-05-16:
Hi. Is someone still interested in fixing this bug?

ehajri, 2021-05-17:
@exelban Hello. Yes I am.

exelban, 2021-06-01:
Hi. Just an update. I do not forget about this problem.
I'm currently working on #454 and it will change the way of how the app changes the fan speed. 
And or it will fix this issue, or give a strict answer of why it does not work.

CharlieJ0hn, 2021-06-04:
thank u

cristiean, 2021-06-11:
Fresh install of Big Sur and I can't control my fans either

exelban, 2021-06-18:
Hi. Could someone check this build?

[Stats.dmg.zip](https://github.com/exelban/stats/files/6677100/Stats.dmg.zip)

cristiean, 2021-06-18:
> Hi. Could someone check this build?
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/6677100/Stats.dmg.zip)

Not working on 13'' 2018 MBP. The only difference is that it asks me for admin password every time I move the slider.

exelban, 2021-06-18:
@cristiean Do you move the app to the application folder?

cristiean, 2021-06-18:
Yes. I've also downloaded the Macs Fan Control app and that one works, so it's not my machine or anything like that.

exelban, 2021-06-18:
@cristiean Hmm. It's very strange.
Could you please run:

```
/Applications/Stats.app/Contents/Resources/smc fan -id 0 -m 1
/Applications/Stats.app/Contents/Resources/smc fan -id 0 -v 3500
```

And check if your fan changes the speed.

exelban, 2021-06-18:
Do not run this commands with sudo.

cristiean, 2021-06-18:
Running without permissions:
[image]

And with super user. The fan speeds haven't changed.
[image]

exelban, 2021-06-18:
@cristiean 
Could you please run: `/Applications/Stats.app/Contents/Resources/smc list`, find the value `F0Tg` and share it.

exelban, 2021-06-18:
Very strange that it does not run without sudo. And more strange that the fan does not change the speed after you run it with sudo.

cristiean, 2021-06-18:
[F0Tg]     1250.0

Also. Not sure whether you're using [smcFanControl](https://github.com/hholtmann/smcFanControl) as part of Stats. I tried smcFanControl earlier and it doesn't change my fan speed. Only Macs Fan Control does (I know-- this one isn't open source).

exelban, 2021-06-18:
Hmm, how it's possible that some of the macs do not properly work with SMC.
`smcFabControl` uses the same way as Stats to work with fans.

exelban, 2021-06-18:
@cristiean could you please check if the output of `/Applications/Stats.app/Contents/Resources/smc list` has some keys with name `F0Md` or `F1Md`?

cristiean, 2021-06-18:
> @cristiean could you please check if the output of `/Applications/Stats.app/Contents/Resources/smc list` has some keys with name `F0Md` or `F1Md`?

It does. They both map to `0` when the fans are off and to `1` when they're running

exelban, 2021-06-18:
Thanks. Could you try this one?
[Stats.dmg.zip](https://github.com/exelban/stats/files/6679399/Stats.dmg.zip)

cristiean, 2021-06-18:
Doesn't work

exelban, 2021-06-18:
Does the value of F0Md change when you switch between auto and manual mode?

cristiean, 2021-06-18:
No. I even moved the slider around. `F0Md` and `F1Md` remain set on `0`

exelban, 2021-06-18:
Hmm.
Could you please try these commands:
```
sudo /Applications/Stats.app/Contents/Resources/smc set -k F0Md -v 01
sudo /Applications/Stats.app/Contents/Resources/smc fan -id 0 -v 3500
```

cristiean, 2021-06-18:
[image]

Although it says that, both `F0Md` and `F1Md` remain `0`

exelban, 2021-06-18:
After `smc set -k F0Md -v 01` the value of `F0Md` in the `smc  list` still 0?

cristiean, 2021-06-18:
yes, just checked again. Even closed Stats and tried again.

exelban, 2021-06-18:
Ok, it looks like I need to have some mac with T2 to figure out how the fan could be controlled. Thanks for your help. I will give to know when I will have more info about this.

cristiean, 2021-06-18:
Cool. The SMC tool seems a bit weird. I try using it myself and some of the options don't even work. e.g. when I try `./smc -f` to list the fans, the help text comes up. Is it the same for you? Is this a known bug in smc? Are you using the correct version?

exelban, 2021-06-18:
SMC (System Management Controller) is a part of macos. There is no option to use the wrong version, or version at all.

cristiean, 2021-06-25:
@exelban I see that you've committed a fix. Is it pushed in version 2.5.20? I can confirm that I'm on 2.5.20 and once I input the admin password, if I quit the app and open it again, the password is remembered. Fan control is still not working, though.

exelban, 2021-06-25:
@cristiean Hi. the fix if waiting for the next release. It will be on Sunday.

exelban, 2021-06-25:
Ok. There is a debug build. You can use it till Sunday)
[Stats.dmg.zip](https://github.com/exelban/stats/files/6717635/Stats.dmg.zip)

cristiean, 2021-06-25:
@exelban thanks! Just tested it now and it works!! Very well done!! 👏 👍 Just a UX thought: Would it be possible to set the fans back to "Automatic" when quitting Stats? Apart from that, well done! What was the "problem"? How did you get it working for the T2 chip macs?

exelban, 2021-06-25:
Yes, it's possible. But it needs to be realized as an option in the settings. Because someone would like to keep it always in the manual mode.

On macs with T2 SMC fan keys has a different type than on macs without T2.

closed 2021-06-27 by exelban