exelban/stats #432: [Bug] Widgets color reset after update.

lucasaf04, 2021-04-20:
**Describe the bug**
Widgets colors have been reset after update to latest version. After setting them back, GPU % widget color isn't saved and reverts back to system accent if you close and reopen Stats.

**Details:**
 - Device: Macbook Pro M1
 - MacOS 11.2.3
 - Application version 2.5.10

exelban, 2021-04-20:
Hi.
It's was because I was added a translation to the colors. And must change the keys. Sorry for not include this information in the release notes.

lucasaf04, 2021-04-20:
@exelban But GPU % widget shouldn’t save color value? As I mentioned it resets it to system accent, while the rest of the widgets work properly. Sorry if I didn’t explain well.

exelban, 2021-04-20:
Oh, sorry. Don't understand the problem with GPU.
I have rechecked it now and everything looks ok. What exactly widget have this problem?

lucasaf04, 2021-04-21:
@exelban This one: 
[image]

exelban, 2021-04-21:
Hmm. It's not saving the color when you change to something else?

lucasaf04, 2021-04-21:
@exelban Yes.

exelban, 2021-04-21:
Could you please run `defaults delete eu.exelban.Stats` and check if it helps?

lucasaf04, 2021-04-21:
@exelban It doesn’t help.😢 After setting up everything again, now gpu % widget defaults to monochrome accent, the rest works fine.

exelban, 2021-04-21:
I check this behavior on 2 macs. And it works. 
Does the color not change when you change it in the settings? Or it resets after restart?

lucasaf04, 2021-04-21:
It resets after restart. I have also checked it on my intel Mac and the behavior is the same.

exelban, 2021-04-21:
What happens when you select red color?

exelban, 2021-04-21:
What do you mean by `same`?

exelban, 2021-04-21:
The color of the value in the widget does not change from black to red?

lucasaf04, 2021-04-21:
It changes, I close and reopen stats and it’s on monochrome accent

exelban, 2021-04-21:
Are you sure that it happens only with GPU?

lucasaf04, 2021-04-21:
Yes

exelban, 2021-04-21:
Very strange. Because there is one "sharable" widget.
Could you make next steps:
- run `defaults read eu.exelban.Stats GPU_mini_color` and check the output
- change the color and run `defaults read eu.exelban.Stats GPU_mini_color` again
- restart the app and run `defaults read eu.exelban.Stats GPU_mini_color`

Post please the output of each run.

lucasaf04, 2021-04-21:
`defaults read eu.exelban.Stats GPU_mini_color
2021-04-21 21:37:23.189 defaults[56248:208517]
The domain/default pair of (eu.exelban.Stats, GPU_mini_color) does not exist`

lucasaf04, 2021-04-21:
If I check iGPU_mini_color:
First run: app just opened, showing monochrome accent.
Second run: color changed to green.
Third run: app restarted and showing monochrome accent.
[image]

exelban, 2021-04-21:
Ok, I found the problem. If you disable the `Show GPU type` options, everything will work fine.

exelban, 2021-04-21:
It will be fixed in the next release. 
Thanks for the feedback and help with debugging the problem)

arisudesu, 2021-04-22:
Also, the "second purple" color suddenly became black

exelban, 2021-04-23:
@arisudesu Hi. Thanks for reporting. I have noticed this, and fix will be today.

closed 2021-04-25 by exelban