exelban/stats #434: Possible to have more than one HDD on watchlist at nav bar?

phattrance, 2021-04-23:
Current app only allows us to view only one HDD on the navbar on big sur. Is it possible to have more options so that we can perhaps add 2 or more drives in the navbar?

[image]

exelban, 2021-04-26:
Hi. Thanks for the proposal.
Unfortunately there is no option to do that now. And I'm not sure that it will be added in the future.
The problem not to read a few disk stats at the same time (it already working like this). But to visualize a few disks at the same time.

closed 2021-05-02 by exelban

dannybeckett, 2025-05-19:
@exelban Is it possible to visualize this now? I would really love this feature, it's the one thing that is missing. To be able to see the % used of each disk on the navbar would be great. Even to have just 2 would be great! Thanks a lot