exelban/stats #435: Feature request: change font

gelpenaddict, 2021-04-23:
The font looks tiny especially on WQHD+ monitors. Add an option in settings to change the font.

exelban, 2021-04-24:
Hi.
No, there will be no option to change the font. This feature could provide a lot of bugs and unexpected behavior.

closed 2021-04-24 by exelban

gelpenaddict, 2021-04-24:
How about an option to separate up/down so it won’t use half the available height? Unusable the way it is now.

exelban, 2021-04-24:
Read this one: #420

gelpenaddict, 2021-04-24:
Read it.

Have the option to show download only, or separate items for down vs up.

Max 3 digits + unit, ie 123 MB/s, 123 KB/s, making it constant width.

gelpenaddict, 2021-04-25:
I haven't seen any other app split a menubar space into top/bottom for text. 

For example, the clock/calendar in the upper right corner puts date and time side by side (if you enable both).

I think the choice of putting up/down speed in the same space is akin to putting date and time on top of one another, which Apple does not do.

Not sure why it's such a big ask to be more consistent with the rest of MacOS UI.