exelban/stats #439: network module: local IP shown as "Unknown"

meiser79, 2021-04-26:
Hi,

the network module shows the local IP as "Unknown".

ifconfig output of my wifi interface looks like this (IPv6 address anonymized):

en0: flags=8963<UP,BROADCAST,SMART,RUNNING,PROMISC,SIMPLEX,MULTICAST> mtu 1500
	options=400<CHANNEL_IO>
	ether 14:7d:da:a9:db:a2 
	inet6 fe80::da:9be5:8abe:3423%en0 prefixlen 64 secured scopeid 0x6 
	inet 192.168.1.8 netmask 0xffffff00 broadcast 192.168.1.255
	inet6 3b02:518:c010:3110:856:b38f:14d5:1586 prefixlen 64 autoconf secured 
	inet6 3b02:518:c010:3110:adb6:3820:4d75:23e0 prefixlen 64 autoconf temporary 
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active

**Details:**
 - Device: Macbook Pro 2020
 - MacOS 11.2.3
 - Application version 2.5.11

Thanks for having a look!

Kind Regards,
meiser

exelban, 2021-04-26:
Hi. Thanks for reporting.
Does it always shows `Unknown`? Or after the restart, it shows an IP?

meiser79, 2021-04-26:
Hi, it always shows "Unknown". A restart doesn't change it.

exelban, 2021-04-26:
Hmm, strange. Do you have selected the correct interface?

meiser79, 2021-04-26:
OK, found out why it's not shown. Because of #296, I use the "process based" setting to show the throughput. As soon as I switch to "interface based", the local IP address is shown.
Any chance to show the local IP address even with "process based"?

exelban, 2021-04-26:
Your mac could be connected to few networks via different interfaces. And when you do not choose the interfaces. I could not show the correct IP address.

meiser79, 2021-04-26:
It does work here when selecting "interface based" and "autodetection". So I thought, it's just a matter to make it work with "process based". But if it's not possible, you can just close the issue.

exelban, 2021-04-26:
The difference between interface-based and process-based it's in the method of how stats are collecting.
When `interface-based`, Stats find active interfaces and "watching" the data on the interface. And having the active interface, I know the local IP of this interface.
When `process-based`, data are collecting from the `top`. And Stats does not know what interface is using.

So when `process-based` is selected, there is no interface. And it's the reason why it's impossible to detect the interesting interface.

closed 2021-04-26 by exelban