exelban/stats #441: Feature request: Show kernel CPU usage in menu

nikodiyana, 2021-04-27:
One cool feature of MenuMeters is showing the kernel CPU usage AND app CPU usage in the actual menu gauge.

Would be great to implement this!

Keep up the great work.

exelban, 2021-04-27:
Hi. Thanks for the proposal.
Could you please tell a little bit more about this feature?

nikodiyana, 2021-04-28:
It is the same what you already implemented in the CPU pop-up window: System usage (in red) vs User usage (in blue). It would be very useful to show these colours also in the menu, in the vertical bar widget. (The pie already has it).

exelban, 2021-04-28:
Do you propose to visualize the system and user usage in the bar chart?

nikodiyana, 2021-04-28:
Yes

exelban, 2021-04-28:
There is a pie chart for this. Bar chart will show only 1 value.

nikodiyana, 2021-04-29:
Sounds like definitive no on a feature request having the bar chart showing system CPU usage.

The need for this feature:

Users like me set different type of charts for different sensors. This way you can save menu space by hiding the labels. 

In my case RAM is pie chart, GPU is single bar chart, CPU is multi-bar chart. The system state is perfectly clear from a quick glimpse at the menu. Yet, we can't know if kernel is busy or app is taking the CPU at the moment.

I hope it is not difficult to implement.

exelban, 2021-04-29:
Ok, I will take a look at this.

nikodiyana, 2021-04-29:
That's great, thanks!

closed 2021-08-09 by exelban

Tai-Zhou, 2021-08-14:
Thanks for the update! I have a minor suggestion that the setting of "split values" should be put into the settings of bar charts, or would this feature also be brought to the line charts in the future?

exelban, 2021-08-14:
@Tai-Zhou this option will be available in the settings only if you have enabled the bar chart. And it must be in the settings of the module, not a widget.

Tai-Zhou, 2021-08-15:
> @Tai-Zhou this option is available only if you have enabled the bar chart. And it must be in the settings of the module, not a widget.

I see. It seems the option is always visible in my RAM setting page, while it only appears with the bar chart in the CPU setting page.

[image]

exelban, 2021-08-16:
@Tai-Zhou fixed