exelban/stats #442: Issues in macOS 11.3

RoyRao2333, 2021-04-27:
I just updated to macOS 11.3 and I found that the Battery indicator become black and options such as _Additional information_ and _Colorize_ don't take effect.
[image]

exelban, 2021-04-27:
Hi. Thanks for reporting.
I will take a look.

RoyRao2333, 2021-04-27:
> Hi. Thanks for reporting.
> I will take a look.

And when I add a new status item, its icon will become transparent (invisible).
Waiting for the hot fix. :)

exelban, 2021-04-27:
Ok, I update my mac to 11.3. And everything works fine.
The battery in the popup view will be colorized when the level is different than 100%.

Could you explain what do you mean by `icon will become transparent (invisible)`?

exelban, 2021-04-27:
Also, additional information visible only when available. Again when battery not fully charged.

closed 2021-04-27 by exelban

RoyRao2333, 2021-04-28:
> Ok, I update my mac to 11.3. And everything works fine.
> The battery in the popup view will be colorized when the level is different than 100%.
> 
> Could you explain what do you mean by `icon will become transparent (invisible)`?

Hi! The thing is, for example, when I turn Network switch on, and with other options off, I got an invisible button...
[image]
[image]