exelban/stats #443: Invisible status item and the change of battery icon

RoyRao2333, 2021-04-28:
**Describe the bug**
Hi! So according to my previous report, for example, I turned Network switch on and leave the rest options off, then I got an "invisible" button like below:

[image]
[image]

I thought even if I turn off options below, it should have a default icon to remind users that they have turned on Network status.

And furthermore, I noticed that when I plug in my power adapter, the battery icon changes to *Charging* style in about 2~3 seconds. Maybe it should change within 1 sec? Could this be a bug? I'd appreciate it if you check this out. :)

**Details:**
 - Device: Macbook Pro 2017
 - macOS 11.3
 - Application version 2.5.11

exelban, 2021-04-28:
Hi. Thanks for reporting.
And much better description than in the last issue.

Network widget. There are different options of how you can customize widgets. And in this case, if you will disable pictogram and value (widget could contain only these parts). You will receive an empty widget. There are default values, when you changing it by disabling value or pictogram, you will receive an empty widget.

Battery. When you connecting the power adapter to the mac. The charging will not start at the same moment when you connect it. There is even some lag to switch between battery power to the power adapter. It's normal that mac will start charging after some period of time (1-10s). And it's normal that mac will start powering from the power adapter, not from the battery (1-3s).

RoyRao2333, 2021-04-29:
> Hi. Thanks for reporting.
> And much better description than in the last issue.
>
> Network widget. There are different options of how you can customize widgets. And in this case, if you will disable [...]

Thanks for the explanations!

closed 2021-04-29 by exelban