exelban/stats #445: Invalid argument messages when clicking on CPU widgets

colebob9, 2021-04-28:
**Describe the bug**
In the output box in Xcode, this message (see screenshot) gets spammed whenever I click on a CPU widget. It also gets output when starting up, right as the CPU widget is loaded. Doesn't seem to affect much else as far as I can tell. I'm guessing it's a M1 specific issue as the message originates from `SMC.swift` and M1 doesn't have SMC. Mostly reporting to make sure that it isn't a bigger issue.

**Details:**
 - Device: M1 Mac Mini, 2020
 - MacOS 11.2.3
 - Application version: latest commit as of writing ac4d816ab37334e4e0b63a841e0cb8742544cb30

[image]

exelban, 2021-04-28:
I think it tries to read the CPU temperature. Thanks. I will escape this for m1.

exelban, 2021-04-28:
@colebob9 could you maybe help with #414?)

closed 2021-05-02 by exelban